Altium-Designer-addons/scripts-libraries #12: AddWireStubsSch-Form.pas

Clements-Peter, 2016-10-28:
Dear All,

I'm using AD16 and tried to make this script work with the given project.
I obtain no error output message and nothing is modified in the schematic document.

Do I need to do something specific except selecting the schematic and click on the “Add Stubs” button because I do not obtain an error message :(

Thanks

andreatta, 2016-10-31:
Hi Clements-Peter,

I updated the project and added some dialogs to inform about failed operations.
Please test again and let me know if you have more success.

It does work here™ ;-)

Best regards

justinmassiot, 2017-04-24:
With no feedbacks since October 2016 I consider this issue to be resolved. Thanks @andreatta for your update.

closed 2017-04-24 by justinmassiot