Altium-Designer-addons/scripts-libraries #17: Clean-up of the repository

justinmassiot, 2018-06-11:
Hi all @Altium-Designer-addons/project-committers,

The repository may need a small clean-up, tell me what you think about the following ideas:

1. Don't keep the Google Drive alive
  a. Not every contributor upload a Zip file to the Drive along with its script update to GitHub -> the one and only up-to-date source is GitHub
  b. [DownGit](https://minhaskamal.github.io/DownGit/#/home) is a perfect replacement for the Google Drive
2. Remove the "#Downloads" folder
3. Update the README + description
  a. Remove or update obsolete information (links, way to contribute, obsolete GDrive information)
  b. Re-order the list of scripts alphabetically (the chronological order isn't needed any more, dates are shown right next to each folder)
  c. Add a download link (based on DownGit) to each element of the scripts list

lunarplasma, 2018-07-23:
If DownGit is used to provide individual download links for each script, we should also have a wiki page entry to describe the process of creating a link in this way, and where to place it in the script directory. This is just to make it easier for developers to add new entries quickly.

justinmassiot, 2018-07-23:
Of course we can create a wiki page for that purpose. Here are some example links:
* Your latest script -> https://minhaskamal.github.io/DownGit/#/home?url=https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/GetPinData
* More complex project -> https://minhaskamal.github.io/DownGit/#/home?url=https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/XIA_Release_Manager

The link can point either to a single file or to a complete folder.

justinmassiot, 2018-09-20:
Work in progress (for information)