Altium-Designer-addons/scripts-libraries #19: Fillet script doesn't work for any-angle tracks (fix hacked together)

RnDMonkey, 2018-09-13:
I recently have had occasion to do a LOT of fillets of specific radius on a mechanical layer, and tweaked the Fillet_v1.3 script (https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Fillet) to do fixed radius fillets and also to work correctly on tracks at any angle to each other. There are a couple safety checks disabled (like you can specify too large a radius to fit the existing tracks), and the undo functionality is completely broken (it seems partially broken before and I've only made it worse and don't feel like sinking more time into it right now). Also, I haven't cleaned up disabled code or made things pretty. Petar, are you interested in my revised code?

justinmassiot, 2018-09-14:
Hello @RnDMonkey, this would be really nice to integrate your updates into the project. Could you please do the following:

1. Fork the project
2. Edit your fork with your improvements
3. Create a Pull request for a re-integration of your fork into the main project

Then it would be easy to check your code and track the modifications against the current script version.
Thank you very much :wink:

justinmassiot, 2018-09-16:
Your [pull request is now merged](https://github.com/Altium-Designer-addons/scripts-libraries/pull/20).

closed 2018-09-16 by justinmassiot