Altium-Designer-addons/scripts-libraries #28: Altium 18 and 19

filmi04, 2018-10-31:
Hello
This script does not work on Altium 18 and 19.
"Undeclared identifier:IsStringANum"
Can this be fixed?

lunarplasma, 2018-10-31:
Can you specify the script/file in question?

filmi04, 2018-10-31:
hi
Fillet and Fillet with Radius do not work

justinmassiot, 2018-11-06:
Hey @filmi04 welcome to our Altium Designer add-ons project.
Do you have some programming skills? I didn't write those scripts myself and must say I don't have much time right now to look at it. And there is not many active contributors left here, except @lunarplasma of course :smile:

lunarplasma, 2018-11-07:
@justinmassiot Unfortunately I don't think I can help so much, as I don't have those versions of Altium! >.<

filmi04, 2018-11-07:
Bad news. We'll have to keep on hand AD17 for these purposes. Thanks for
the answer. By the way, the last update of the Fillet script does not work
in version 17 too.

lunarplasma, 2018-11-09:
@filmi04 You can probably have a go at it fixing it yourself, maybe.
So I had a quick look at the code - the function you mentioned `IsStringANum` exists in FilletWithRadius.pas. It doesn't exist at all in Fillet.pas, so I don't know why that script would be failing for you.

* Is the script you use in AD17 the same script you use in AD18/19?

filmi04, 2018-11-09:
https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Fillet
This script does not work for me in any of the Altium versions. (AD9,
AD17-AD19)
The problem occurs at the stage when it is necessary to enter the value of
rounding.
[image: image.png]

RnDMonkey, 2018-11-09:
Try installing the Fillet With Radius script alongside Fillet script. I
suspect I mistakenly called another script's function without realizing it
wasn't built-in to Altium. If that is the case, I'm surprised it was able
to link to another script in the first place, but that might fix it.

justinmassiot, 2018-11-09:
Thanks @RnDMonkey .
@filmi04 tell us if this is OK and if we can integrate a fix in the master branch.

RnDMonkey, 2018-11-10:
@filmi04 @justinmassiot @lunarplasma fix has been implemented and merged with master. If you are able to verify fix, comment back and we can close issue.

filmi04, 2018-11-12:
Still a problem with this script
https://www.youtube.com/watch?v=zi2cYxhfr5I

RnDMonkey, 2018-11-12:
Should be fixed now. Please verify.

filmi04, 2018-11-13:
Work!
But there is a small bug.
If you immediately select the percentage values - an error!
https://www.youtube.com/watch?v=yD1E208aS2Q

filmi04, 2018-11-14:
In the Fillet.dfm file changed row
Text = '32.25' to Text = '50'
And it all worked. Thank!

justinmassiot, 2018-11-16:
@RnDMonkey, are the modifications all included in the master source code? Could we close the ticket?
@filmi04 did you test with Altium Designer 19?

filmi04, 2018-11-16:
Yes. It's works on 17-19 versions of Altium
Thanks

closed 2018-11-16 by filmi04