Altium-Designer-addons/scripts-libraries #41: LayerSet definition

githubusr2, 2020-02-07:
Hi, I run stitchingvias.pas and get error: Undeclared identifier: LayerSet
I read the file stitchingvias.pas and dont find its definition, even var Board has its definition, but var LayerSet has no defintion, I add its definition like LayerSet : IPCB_LayerSet, but new error appear like:
undeclared identifier: CreateLayerSet, so what problem is it? how to fix it? looking forward to reply, thanks.

justinmassiot, 2020-02-07:
Hello githubusr2,
We may ask @pperisin about this because because he's the original author of the script.
Side question: What does this script bring that the core feature "Via Stitching" doesn't do?

githubusr2, 2020-02-07:
@pperisin @justinmassiot  use altium designer software to open titichingvias script project then open any pcb file then run script of titiching vias then a dialog appear then click ok button, then error appear as i wrote in last message, i dont find any defintion LayerSet of var in the file, so i think this is reason

pperisin, 2020-02-07:
Script was made before core altium had stitching vias. Suggestion to use
core Altium feature.

Regardy,
Petar

pet, 7. velj 2020. 10:07 Justin MASSIOT <notifications@github.com> je
napisao:

justinmassiot, 2020-02-07:
Thanks for your inputs Petar. I will add a note in the ReadMe file.

closed 2020-02-07 by justinmassiot

cdplayer, 2024-03-16:
I'm needing a feature like the core via stitching but want to replace all the objects of type 'via' with a cutout or keepout structure to remove copper. I mostly want these to be circular, so I thought maybe I can read the OD of the via and replace the 'set' structure with other entitys other than vias. Think this is possible? It's a bit like a negative 'thieving'

justinmassiot, 2024-03-20:
Hello @cdplayer,
Many things are possible with scripting.

> I can read the OD of the via and replace the 'set' structure with other entitys other than vias
Not sure it would work, but you can try!