Altium-Designer-addons/scripts-libraries #46: Scripting better documentations !

msamy0, 2020-06-21:
I am searching for better Altium's API documentation, but Altium's provided documentation. It is poor and bad sorted. Also it is pretty complex to search within.

justinmassiot, 2020-06-22:
Hello @msamy0 and welcome!
I saw that you wrote me message on LinkedIn, but I prefer to give a "public" answer.

Yes, the Scripting documentation from Altium is poor. In fact, I may say it's getting worse these days. Please look at [my blog post](https://gotomation.info/2020/03/write-first-script-altium-designer/#documentation-for-scripts-developers) where I deal with finding the right documentation. You might try to use the (very old) [PDF version of the API documentation](https://gotomation.info/wp-content/uploads/2020/03/Scripting-system-reference-TR0135-System-Reference.pdf).

I'm sorry I can't help you more. If you have the chance to report the situation to your management, please do so! The community will thank you :wink:

msamy0, 2020-06-23:
thank you, i appreciate you help :)

msamy0, 2020-06-23:
**Sharing my experience with this API for the community, DO NOT INVOLVE !**   
I have done all procedures in the blog post [here](https://gotomation.info/2020/03/write-first-script-altium-designer/#documentation-for-scripts-developers) even before [Justin ](https://github.com/justinmassiot) share it with me as i was working around this since while. I AGREE, it is the only way to work around this poor API documentation. However, it is still not efficient enough. Also the DXP Units which were provided earlier with AD software in the installation directory is no longer available, unless you participate in DXP Developer Beta Program, in which i tried to participate already, but Altium never respond 👎 . Theses Units help in developing scripts in more capabilities IDE like [EMbarcadero](https://www.embarcadero.com/products/delphi). It is also helpful in tracing functions and classes used in case the documentation isn't sufficient (and it is XD ).
What to Do ? 
If the task to be automated it is less priority, I ADVISE you not to involve in this until a pretty good documentation is provided by Altium. If it is a high priority one, try to find an already available example to do the task [here](https://github.com/Altium-Designer-addons/scripts-libraries), however using the examples itself isn't straight forward as you had to work around the documentation also, because some of the functions don not work, even if you are you copying it from a tested example like the issue i talked about [here ](https://github.com/Altium-Designer-addons/scripts-libraries/issues/47).
Finally, i have scripted in APIs like Solidworks and Fusion360 provides. I can say Altium's one is not acceptable at all. DO NOT USE IT !

closed 2020-07-08 by justinmassiot