Altium-Designer-addons/scripts-libraries #47: AutoStepPlacer Doesn't Work

msamy0, 2020-06-23:
I used [this](https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Scripts%20-%20PCB/AutoSTEPplacer) as a reference for me to add step models to footprints automatically.
I used the following lines to import and implement the 3D model :
`        
 
          //ModelDir > string contains the directory of 3d models
          //TempCellsContent[6] > string list contains data retrieved from CSV file earlier, cell [6] contains the step name.
          //ModelName > a string list to store "FindFiles" function output of the .step full path
          //Footprint > is the first and only component in the (PCBLib) file opened on executing script
          //StepModel > is an IPCB_LibraryComponentBody
          //Model > is an IPCB_Model  

          {**** Adding 3d Model ****} 
           FindFiles(ModelDir,TempCellsContent[6],faAnyFile,False,ModelName);
           Iterator := Footprint.GroupIterator_Create;
           Iterator.Addfilter_ObjectSet(MkSet(eComponentBodyObject));
           StepModel := Iterator.FirstPCBObject;          
 
          //Delete current 3D Models
           While StepModel <> Nil Do
           Begin
           Footprint.RemovePCBObject(StepModel);
           StepModel := Iterator.NextPCBObject;
           End;

           StepModel := PCBServer.PCBObjectFactory(eComponentBodyObject,eNoDimension, eCreate_Default);
           Model :=  Stepmodel.ModelFactory_FromFilename(ModelName[0], false);
           StepModel.SetState_FromModel;
           STEPmodel.Model := Model;
           Footprint.AddPCBObject(StepModel);

           FootPrint.GroupIterator_Destroy(Iterator);

     //Finalize code
     ResetParameters;
     AddStringParameter('Action', 'All');
     RunProcess('PCB:Zoom');
     CurrentLib.Board.ViewManager_FullUpdate;
     CurrentLib.LibraryIterator_Destroy(FootprintIterator
`
As not expected XD, the removing line doesn't work as the old models still exist, and no new models added . the script do run and end without errors ! but nothing change.
When I debugged it line by line, I noticed it goes into the while loop and do pass into the RemovePCBObject line, but nothing change in the workspace ! 
even i tried to add a commands from this [link ](https://techdocs.altium.com/display/SCRT/PCB+API+Design+Objects+Interfaces#IPCB_ComponentBody%20Interface) after the RemovePCBObject line. It is not executed because it cannot find the eComponentObeject as i have already remove it in the previous line. SO the Body is theoretically removed in the execution flow, but after finishing script run, nothing is changed at all as i mentioned before.

justinmassiot, 2020-06-24:
Hello @msamy0,
This script was originally developed by @pperisin in January 2012 😲, it's not surprising that it doesn't work in recent versions of Altium Designer.
If you find a solution to fix this script, please share it with us.

msamy0, 2020-06-24:
I have tried every possible autocomplete option provided in the editor to do so, but nothing worked !
The internal structure of the software isn't clear at all. As I mentioned above, I could somehow edit the eComponentBodyObject which is according to my trials and the poor documentation is the surface level or the ancestor of the 3D model itself, which according to AutoStepPlacer Example is a (TModel) object which the documentation didn't say its name at all !
I just could know its functions and properties through the autocomplete in the editor which doesn't help either. 
so in general i couldn't fix it any way. 
I had the same problems in two more things 
1- Saving .pcblib file 
2 - editing polygons (which have the same problem of ancestors top level and elements bottom level like the 3D models)
------------------------
finally i gave up and scripted WINDOWS 10 itself to do these three tasks left ! :D

closed 2020-08-10 by justinmassiot