Altium-Designer-addons/scripts-libraries #49: Adjust Designators 2: Changes overlay layer despite disabled

cortexer, 2020-07-27:
When using the Adjust Designators 2 script on a mechanical layer it also changes the overlay layer, despite the checkbox "modify overlay" is unchecked.
It fails with the latest Altium Designer version 20.1.11, but worked with older versions (not sure at what exact version it got broken).

When opening an old layout, it still works on those components (placed with older Altium versions), but not on components newly imported into the layout. However I could not find a difference in designator properties, which might different and affecting the script behaviour.

cortexer, 2020-07-27:
What I am trying to do is
- copy designators from overlay to mechanical layer pair using the "CopyDesignatorsToMechLayerPair" script.
- align the mechanical designators to the center of the components (as kind of assembly drawing).

Seems it has to do something with auto centering:
When the autopositioning of the overlay designator is set to "center": the mechanical layer designators get aligned properly by the script. (But obviously I do not want to set the overlay designators to "center", which would put them right in the middle of the component).
When the autopositioning is set to "manual": the overlay designators get centered by the script (but they shouldn't), whereas the mechanical layer designators don't get centered (but they should).

justinmassiot, 2020-08-10:
Hello @cortexer ,
It's always nice to have feedbacks from real users, so thank you for that.

Regarding your issue, I've to say I didn't develop this script myself so I've no obvious idea on the reason of the bug. But if you provide some sample PCB file with which I can easily reproduce the bug, I could try to investigate.

cortexer, 2020-08-11:
Here is a small example file.

The designator has been copied from Top Overlay to a mechanical layer (layer 11, assembly drawing top in this example) using the corresponding script.

Now when using the AdjustDesignator2 script on the mechanical layer:

The resistor R115 was imported with an older Altium Designer version, on this component the script still works.
The mechanical layer is adjusted, the overlay layers stays as is.

The LED H1 was imported with Altium 20 and the script somehow fails and adjusts the overlay instead of the selected mechanical layer. I did not yet find out, what makes the difference.
[AdjustDesignatorScriptFails_Example.zip](https://github.com/Altium-Designer-addons/scripts-libraries/files/5056886/AdjustDesignatorScriptFails_Example.zip)

justinmassiot, 2021-04-29:
Hi @cortexer ,
Even with the recent improvements made by @BrettLMiller in commit 24589e512f38874d09384c79d373c2b1a626b030, your issue remains. At least the behaviour of the script differs for designators R115 and H1 (tested in AD 21.3.1).
Brett may help us with this specific issue.

BrettLMiller, 2021-04-30:
I've made a couple of minor changes 38af21d to AdjustDes2 but I've seen AD20 confuse the FP H1 real designator with the mech layer text. Might be a lost cause.
H1 & R115 have different justification settings; in AD17 only multi-line text has these settings/properties.

You might like to try AdjustDes3 (https://github.com/BrettLMiller/Altium-DelphiScripts/tree/master/PCB/AdjustDesignators3), this does not use the real designator to size  or position other strings.

justinmassiot, 2021-05-05:
Thanks to @BrettLMiller , the latest version should (partly?) fix the described issue.

closed 2021-05-05 by justinmassiot