Altium-Designer-addons/scripts-libraries #50: AdjustDesignators2

RainZhangZZFLS, 2020-08-09:
[SimplePCB.zip](https://github.com/Altium-Designer-addons/scripts-libraries/files/5047538/SimplePCB.zip)

Hello Justin,

When I tried the AdjustDesignator2 with this simple PCB design, I found nothing changed and no 'designators' were adjusted. I wonder if you can provide some technical support?
Thanks!

Best,
Runze

justinmassiot, 2020-08-10:
Hey @RainZhangZZFLS ,
Thanks for your feedback.

Which version of Altium Designer do you use please?

RainZhangZZFLS, 2020-08-11:
> Hey @RainZhangZZFLS ,
> Thanks for your feedback.
> 
> Which version of Altium Designer do you use please?

Hey Justin,

Thanks for your reply. My Altium Version is 20.1.11.

justinmassiot, 2020-08-11:
So you may be impacted by the existing issue #49 . I'll investigate both tickets in parallel.

RainZhangZZFLS, 2020-08-11:
Thanks for your reply! Good luck and look forward to your good news!

RainZhangZZFLS, 2020-09-04:
> So you may be impacted by the existing issue #49 . I'll investigate both tickets in parallel.

Hello Justin,

I found the AdjustDesignator works for my PCB. By the way, I wanna ask which commands should I use to adjust the mirror property of the 'Designator'?