Altium-Designer-addons/scripts-libraries #51: RemovePCBObject NOT WORKING !

msamy0, 2020-10-28:
shortly, this function (RemovePCBObject) is not working at all with me 
how could i solve this ? 
tried it on Altium Designer 20 and the oldest supported version (altium 10) , both not working ! 

this method is used in some examples provided Here >> [RemovePCBObject .zip](https://github.com/Altium-Designer-addons/scripts-libraries/files/5455218/RemovePCBObject.zip)

Thanks for support..

justinmassiot, 2020-10-29:
Hi @msamy0 ,
Could you please give more details? I'm not sure what is working, what is not, and how should things work.

msamy0, 2020-10-29:
Ok, sorry for missing details !
what is not working >> **RemovePCBObject** method
How actually it should work >> remove PCB object. In my case i need to remove Tracks from a certain layers.
what happens with me >> Tracks is not removed when i run script 

here's the script which the method is applied in, exactly in line 51 
[DeletePCBObjects.txt](https://github.com/Altium-Designer-addons/scripts-libraries/files/5457706/DeletePCBObjects.txt)

simply the script creates TInterfaceList in which it collects all tracks i need to remove, then run a FOR loop to remove them item by item from the list created. it is doing though (collecting all then delete all, instead of delete them one by one ) because it is what altium recommend in their documentation for this method

https://www.altium.com/ru/documentation/altium-dxp-developer/pcb-api-system-interfaces-reference

[image]

**thanks, I really appreciate your help 🙏**

justinmassiot, 2020-10-30:
Here on GitHub is a project that only _uses_ the scripting API, but you're reporting a bug about the API/engine itself. I'm sorry that I can't help you further.
I suggest you to contact your local support representative from Altium, or to post in the Altium forum.

closed 2020-10-30 by justinmassiot

msamy0, 2020-11-05:
I found a working solution !
--------------------------
i found it [here](https://forum.live.altium.com/#posts/216949/687594) on Altium's forum.
Simply, if you wanna apply this function on a certain footprint in a PCBLib file, then you have to make another footprint in focus !
it is a bug i know, but it it is how things work.
So, **It is impossible** to apply this function in a .pcblib file with only one footprint in,however, in case you have to do though, then create an empty component to just get it in focus while applying the script on the required component.

here's a working code deletes every single primitive in the 1st created footprint in a .pcblib file 
```
procedure RemovePCBObjectSOLUTION;
Var
targetPCBLib : IPCB_Library;
APartIterator : IPCB_LibraryIterator;
AGroupIterator : IPCB_GroupIterator;
APCBLibComp  : IPCB_LibComponent;
APrimitive : IPCB_Primitive;

Begin
targetPCBLib := PCBServer.GetCurrentPCBLibrary;
if targetPcbLib = Nil Then Exit;

//create footprint iterator
APartIterator := targetPcbLib.LibraryIterator_Create;
APartIterator.SetState_FilterAll;

//get the first component, which has primitives needs to be deleted
APCBLibComp := APartIterator.FirstPCBObject;

//get the second component, which is working as a bridge only (as I have no idea how to select component by index ^^ )
APCBLibComp := APartIterator.NextPCBObject;

//make this empty bridge component in focus
targetPCBLib.SetState_CurrentComponent(APCBLibComp);

//then get again the first component to delete from it
APCBLibComp := APartIterator.FirstPCBObject;

//create primitives iterator to loop thru the footprint
AGroupIterator := APCBLibComp.GroupIterator_Create;
AGroupIterator.AddFilter_ObjectSet(AllObjects);
AGroupIterator.AddFilter_AllLayers;
APrimitive     := AGroupIterator.FirstPCBObject;

//delet3 each primitive until the footprint is empty
while APrimitive <> Nil do
begin
   APCBLibComp.RemovePCBObject(APrimitive);
   APCBLibComp.GraphicallyInvalidate;
   APrimitive := AGroupIterator.FirstPCBObject;
end;

End;
End.

```