Altium-Designer-addons/scripts-libraries #53: DownGit does not work

mm1711, 2020-12-30:
Downloading of any script ends with an empty ZIP file

justinmassiot, 2020-12-30:
Hello @mm1711 , could you please provide:

- the version of your Mozilla Firefox browser
- an example URL to DownGit which gives an empty ZIP archive

Thanks.

mm1711, 2020-12-30:
Firefox version is 84.0.1 64bit

example link: https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Scripts - PCB/FixConnections

justinmassiot, 2020-12-30:
> example link:
> https://altium-designer-addons.github.io/DownGit/#/home?url=https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Scripts - PCB/FixConnections
This URL is not a DownGit link, where did you get it?
Here is the correct one: https://altium-designer-addons.github.io/DownGit/#/home?url=https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Scripts+-+PCB/FixConnections (found here: https://github.com/Altium-Designer-addons/scripts-libraries/blob/master/Scripts%20-%20PCB/README.md)

mm1711, 2020-12-30:
On the page https://github.com/Altium-Designer-addons/scripts-libraries/tree/master/Scripts%20-%20PCB click on "FixConnections"
This opens a new page https://altium-designer-addons.github.io/DownGit/#/home?url=https:%2F%2Fgithub.com%2FAltium-Designer-addons%2Fscripts-libraries%2Ftree%2Fmaster%2FScripts%20-%20PCB%2FFixConnections
A download dialog asks what I want to do with FixConnections.zip
FixConnections.zip is empty.
On the mentioned page I can click "Create Download Link" this results in the link from my former comment.
[image]

BTW the link you supplied in your comment results in an empty zip file too

justinmassiot, 2020-12-30:
I have tested with the same Firefox version as yours and this gives me exactly the files I expect after ZIP extraction. Note: I haven't signed in to GitHub.
Same (expected) behavior with Google Chrome 87.0.4280.88 but this time I have signed in.

Are you sure you're antivirus is not in cause?

mm1711, 2020-12-31:
I found the problem.
I use the Firefox NoScript extension. I had enabled almost all github scripts except raw.githubusercontent.com.
This was the problem. After enabling this script it is working now.
Sorry for the inconvinience.

closed 2020-12-31 by mm1711