Altium-Designer-addons/scripts-libraries #54: Cannot commit folder

mm1711, 2021-01-05:
Hi Justin,
this regards to our discussion on https://forum.live.altium.com/#posts/244487/760373.
I installed the library but could not commit a project.
I re-read the readme and found that the project has to be under version control already.
As in my forum post mentioned AD21.0.8 cannot add sub folders.
Now I tried it this way:
1.  Add folder to version control
2. Reports some errors regarding sub folders
3. All files have a blue plus sign
4. Commit whole project => reports same errors
5. All files have still a blue plus sign
6. Now Commit project with your script
7. All files are added and commited but in AD all files have still a blue plus sign

justinmassiot, 2021-01-05:
Okay, I may have read your initial post too fast and I haven't noticed that your project was not already under version control. Sorry for that.

Altium Designer is (apparently) not "able" to manage documents against a version control server if you haven't previously configured the project through the AD GUI. Which is a pain, by the way.
I personally never use this feature because I always add files to version control thanks to TortoiseSVN through the Windows explorer. Then AD become aware of the VC status of the files and I can use the SVN client I want to interact with them: either Altium Designer, TortoiseSVN, or TortoiseSVN through AD thanks to my script.

Regarding your point no. 7, did you try a refresh? Right click, Version Control, Refresh.

mm1711, 2021-01-05:
Point #7: Refresh was the trick

closed 2021-04-02 by justinmassiot