Altium-Designer-addons/scripts-libraries #55: CopyDesignatorsToMechLayerPair

IanGrantham, 2021-01-22:
[image]

When trying to run this script I'm constantly getting this error and I'm unsure what I need to do. Sorry new to coding

justinmassiot, 2021-01-25:
Hi @IanGrantham ,
Thanks for your report.

When you file an issue, please try to do the following:
- Tell us your version of Altium Designer
- Give the name of the script you're trying to use
- Explain the steps to reproduce the error: which context, type of file, what do you expect, etc.
- If you want to point out some portions of code, please use the "Insert code" button

That being said, I think that the `LayerStack` API's backward compatibility has recently been broken. The script you use probably needs some update to work with the latest versions of AD, which may explain the error you encounter.
If I have some spare time in the coming months - which is not guaranteed - I might try to fix this. Or you can dig into the documentation and make experiments on your side to try to fix this script, then give back the changes to the community.

Have a good day.

justinmassiot, 2021-04-21:
This issue should be closed thanks to the additions done to the script by @BrettLMiller in PR #60.
Tested by myself in AD 21.3.0 (beta) => worked as expected.

closed 2021-04-21 by justinmassiot