Altium-Designer-addons/scripts-libraries #56: ZIP for Menu_SVN and rename DXPPrf is not working- UI Customizations

GoogleAman, 2021-03-15:
HI ,
I have follow the below steps but it is not working for load preferences :
1) Make a ZIP archive file containing the files `DXP.RCS` and `IncludedFiles.ini`
2) Rename the ZIP file in `Menu_custom.DXPPrf`
3) Open Altium Designer and load the file `Menu_custom.DXPPrf` from the preferences panel  

The DXPPrf are not loading preference screen to select customizations , Downloaded file is working but when i unzip and zip same file its not working.

Thanks
Aman

GoogleAman, 2021-03-16:
Any help in this regard will be appreciated

justinmassiot, 2021-04-02:
Hey @GoogleAman , my apologize for the delay.
I'm glad to see that my menu got your attention 🙂

Could you please be as precise as possible in your description of your problem:
- Which version of Altium Designer do you use?
- You stated this: `Downloaded file is working but when i unzip and zip same file its not working.` => Just to be sure:
    - What is the "downloaded file"?
    - Why do you unzip then zip again?
    - What is "not working"?

I've just run the test with AD 21.2.2 following [the procedure described in the wiki](https://github.com/Altium-Designer-addons/scripts-libraries/blob/master/UI%20customizations/Menu_SVN/README.md) (the one you seem to have followed) and it does work: the selection dialog appears.

justinmassiot, 2021-04-21:
@GoogleAman is your issue still unsolved?

closed 2021-04-29 by justinmassiot