Altium-Designer-addons/scripts-libraries #59: Track for obsolete scripts

justinmassiot, 2021-04-17:
1. Test each script with a recent version of Altium Designer,
2. Determine if it's running,
3. If not, decide if it's worth trying to upgrade the code:  
  a. Has the feature been integrated by Altium inside the core software?  
  b. If not, does the upgrade represent a small programming effort?
4. If not, move the "old and unmaintained" script into a folder named `_LEGACY_` or equivalent.

ZeroCommits, 2022-10-12:
Hello @justinmassiot, were you able to create a list of obsolete scripts?

justinmassiot, 2022-10-12:
Hello,
Unfortunately no. I had no free time these months and this action is time consuming.
Would you like to help?

ZeroCommits, 2022-10-12:
I probably cannot help going over all of the scripts as some I do not have a use case for, but I will post here what works and what doesn't.

justinmassiot, 2023-06-30:
Dear contributors @RnDMonkey , @BrettLMiller , @coffeenmusic , @CoreyBeyer , @zharovdv ,

Could you think of a way to identify scripts which are "current" (still working in the latest version of Altium Designer, and not superseded by an equivalent core feature) from those which are not? Should we manage some sort of "Archive" folder to not mix current and old scripts together?

Take "Scripts - PCB/CopyDesignatorsToMechLayerPair" for example. To me, "Add Designators for Assembly Drawing" does the same.
Then, should we keep the script?

Justin

zharovdv, 2023-06-30:
> Take "Scripts - PCB/CopyDesignatorsToMechLayerPair" for example. To me, "Add Designators for Assembly Drawing" does the same. Then, should we keep the script?

Should keep of couse. This repository not only for users, it's for developers too. It contains real life altium script examples with many workarounds, tips and tricks.

justinmassiot, 2023-06-30:
> Should keep of couse. This repository not only for users, it's for developers too. It contains real life altium script examples with many workarounds, tips and tricks.

Don't worry, the question was not if I need to delete them or not.
The question was: could we put them aside, or mark them as "unmaintained"?

CoreyBeyer, 2023-06-30:
> Take "Scripts - PCB/CopyDesignatorsToMechLayerPair" for example. To me, "Add Designators for Assembly Drawing" does the same.

For situations like this one, there will likely be some cases where users prefer a customized script which works slightly different from native functionality which could arguably supersede the script. For this reason, I'd argue against moving them into an archive. There could however be something said about marking some scripts as "unmaintained" as @justinmassiot mentioned.

RnDMonkey, 2023-07-25:
> For situations like this one, there will likely be some cases where users prefer a customized script which works slightly different from native functionality which could arguably supersede the script.

I agree with Corey. For this specific example, the built-in Altium feature has the following constraints that may not be desirable:

1. Designators are only added to mech layers of type "Designator", whereas users may [still] be using layers of type "Assembly" or other
2. It's all-or-nothing. If you use the tool, all designators are rebuilt from scratch
3. Format is fixed so user has to reformat everything to customize

Other than that, I agree that there should be an "Unmaintained" section and tag for scripts, but it's a lot of work to determine which those are.

justinmassiot, 2023-07-27:
What about renaming the containing folder of each script that hasn't been modified in the last 2 years? (Maybe 3 or 5 years?)
Adding a suffix like " (unmaintained").

Side question: do you guys know how to check the current version of Altium Designer from inside a script? Is it a trick you use to change the behavior of a script accordingly?
@Altium-Designer-addons/project-committers + @coffeenmusic + @BrettLMiller + @igorpnk + @DTJ40 + @zharovdv

zharovdv, 2023-07-27:
> Side question: do you guys know how to check the current version of Altium Designer from inside a script? Is it a trick you use to change the behavior of a script accordingly?

```
ShowMessage(GetBuildNumberPart(Client.GetProductVersion, 0));  // 23
ShowMessage(Client.GetProductVersion);  // 23.0.1.38
```