Altium-Designer-addons/scripts-libraries #69: AdjustDesignators2

IanGrantham, 2022-05-27:
Hello and thank you for these scripts they are a really big help. I do have one issue and I do not have any experience in coding. The issue I have is when I want to adjust mechanical designators that I have adding from another script from here (CopyDesignatorsToMechLayerPair) I find it is changing the silk on the top overlay as well. From looking a little at the script I belive it uses the overlay designator to position the mechanical .designator and then after it is meant to reposition the overlay back to its original position. What I finding is its not repostioning back to exaxtly its original position and is actually off. I have obvisual not got the tick boxes checked for the overlay as this is not want I want adjusting.

The altium version I'm using is 20.2.8

antifar, 2022-05-27:
邮件已收到，祝好。^_^

antifar, 2022-07-06:
邮件已收到，祝好。^_^

BrettLMiller, 2022-08-09:
There is a later version Sept 2021 here : https://github.com/BrettLMiller/scripts-libraries/tree/AdjustDesignator/Scripts%20-%20PCB/AdjustDesignators2
This is to correct a small offset caused by Multiline.

RnDMonkey, 2022-08-09:
I've pulled Brett's fix in, which I believe addresses this issue.

closed 2022-08-09 by RnDMonkey