Altium-Designer-addons/scripts-libraries #200: AssemblyTextPrep not working with a single sided board

justinmassiot, 2024-10-30:
Hello @RnDMonkey ,

I'm currently testing your script 'AssemblyTextPrep' on one of our projects but I think I've found a little bug in it.

The test is on a board with components on Top layer only.
The PCB is already populated with ".Designator" texts on Mech layer 25 because, in my setup, M25/M26 is the pair used to put RefDes (function "Designator").
[image]

At starts, the script pops an error: "Couldn''t determine top assembly layers from existing components or mech layer pairs:".

After a quick look into the code, it looks like the `MLP.PairDefined(ASSY_LAYER_BOT, MLS.GetMechanicalLayer(i).LayerID)` function doesn't do what we expect from it:

1. 'ASSY_LAYER_TOP' is correctly set
2. The script looks for a corresponding 'ASSY_LAYER_BOT'
3. But it doesn't find any (`ASSY_LAYER_BOT := MLS.GetMechanicalLayer(i).LayerID;` is never reached)
4. And ends with an error

I wonder how it's even possible!
Do you have any idea to fix this?

As soon as I put a component on the Bottom side, the script starts correctly.

--

Side note: the name of the Mech layers are reported incorrectly:
[image]

In my case it should have been "**Mechanical** Layer **25**", instead of "Inner Layer 24".

RnDMonkey, 2025-08-22:
Somehow I missed this issue until Brett commented on it and I got an email. 

I'd guess the issue is that it can't find an existing .Designator text on bottom layer to use for inferring the other layer in the pair. I'd anticipate having bottom-only assembly would have the same issue. It _could_ assume that the other layer is an adjacent layer but that's not very robust.

@BrettLMiller if you've identified the issue, do you want to do a PR? Either Justin or I can review and merge into main. Thank you for noticing this.

BrettLMiller, 2025-08-24:
I've pushed a potential fix here [https://github.com/BrettLMiller/scripts-libraries/blob/AssemblyTextPrep/Scripts%20-%20PCB/AssemblyTextPrep/AssemblyTextPrep.pas].
The script works in AD17 if you avoid invoking the StatusbarManager code.

Have tested using:
- AD17 (only).
- designators on ASSY layer pair of mech 31 & 32.
- forced different code paths to test GetMechLayerObject() usage by presetting ASSY_LAYER_* := eNoLayer etc.
The code in function GetMechLayerObject() has been used in AD17-AD24 in other scripts.

BrettLMiller, 2025-08-24:
Committed update with improved iterator layer filters that support all layers.
I did not change the layer filters hard coded for component placement layers.
Changed ".Designator" string value test use SameString() fn with ignore case.

Tested basic operation with AD22 & AD17.
AD22 test used .Designator strings on mechanical layers 1023 & 1024.

closed 2025-08-26 by RnDMonkey